mrWheel/YAPP_Box #4: question: way to adjust tolerance a bit on the ridge and pcb holes

skarfacegc, 2022-02-12:
The pcb stand holes and the lid/base ridge are both super tight for me. Will likely get better when I go back to my .4 nozzle, but I couldn't really figure out a way to add a little bit of extra play on those dimensions.

mrWheel, 2022-02-12:
Which dimension are you referring to.
You can configure everything to your liking …

skarfacegc, 2022-02-12:
the gap between the lid and the base ridge.  Lid doesn't fit easily on base (again, I'm using a .6 nozzle so it's a little rougher than if I was using a .4)

 the inner diameter of the hole on the PCB supports (would like it to be a bit bigger than pinDiameter, while leaving pinDiameter the same).  Can't really make the pin smaller, would like to make the fit a bit easier.

mrWheel, 2022-02-12:
See [this](https://github.com/mrWheel/YAPP_Box/blob/da09d79ad6241897d233f070697c742d5b793dc5/library/YAPPgenerator_v13.scad#L1473) line

you can add or subtract a small fraction (say 0.05 or so) to either the base or the lid.

closed 2022-02-12 by mrWheel