mrWheel/YAPP_Box #5: Variable scope?

udif, 2022-03-04:
I'm working on a new box for the [Colorlight 5A-75B](https://www.colorlight-led.com/product/colorlight-5a-75b-led-display-receiving-card.html) board (please see https://github.com/udif/YAPP_Box/commit/b03609a1828c16700240dd905bf0e89d84e1172a).

While designing the box, I wrote something like:
```
include <./library/YAPPgenerator_v13.scad>
...
pcbLength         = 143.64; // i(5.6496);
pcbWidth          = 91.69; // i(3.6);
...
holeX = (pcbLength - 135.26) / 2;
holeY1 = (pcbWidth - 81.28) / 2;
holeY2 = (pcbWidth - 60.33) / 2;
pcbStands = [
                 [            holeX, pcbWidth - holeY2, yappBoth, yappPin]         // back-left
               , [            holeX,            holeY2, yappBoth, yappPin] // back-right
               , [pcbLength - holeX, pcbWidth - holeY1, yappBoth, yappPin]         // back-left
               , [pcbLength - holeX,            holeY1, yappBoth, yappPin] // back-right
             ];
```
This brings tons of errors like `Ignoring unknown variable 'holeX'`, `Ignoring unknown variable 'holeY1'`, `Ignoring unknown variable 'holeY2'` pointing to the `pcbStands` lines above.
It also brings a bunch of `undefined operation (number - undefined)` but if we understand that somehow `holeX` , `holeY1`, `holeY2` are somehow underfined at that point, then this secondary warning is understandable.
As this leads to non-working standoffs, I found a workaround, by assigning the 5 constants above the YAPP code:

```
pcbLength=0;
pcbWidth=0;
holeX=0;
holeY1=0;
holeY2=0;

include <./library/YAPPgenerator_v13.scad>
...
pcbLength         = 143.64; // i(5.6496);
pcbWidth          = 91.69; // i(3.6);
...
holeX = (pcbLength - 135.26) / 2;
holeY1 = (pcbWidth - 81.28) / 2;
holeY2 = (pcbWidth - 60.33) / 2;
pcbStands = [
                 [            holeX, pcbWidth - holeY2, yappBoth, yappPin]         // back-left
               , [            holeX,            holeY2, yappBoth, yappPin] // back-right
               , [pcbLength - holeX, pcbWidth - holeY1, yappBoth, yappPin]         // back-left
               , [pcbLength - holeX,            holeY1, yappBoth, yappPin] // back-right
             ];
```
The code now works, but I don't understand why was this necessary.

In addition, several new warnings on YAPPgenerator_v13 itself (I'm not giving line numbers, because my fork added a few lines for the rotated cutout code):

```
function getMinRad(p1, wall) = (p1<=wall) ? 1 : p1-wall;
```
(4 times `undefined operation (number <= undefined)` and 4 times `undefined operation (number - undefined)`) which I totally don't understand why.

```
                translate([shellLength-15, -15, 0])
                  linear_extrude(1) 
                    mirror(1,0,0)
                      %text("LEFT"
                            , font="Liberation Mono:style=bold"
                            , size=8
                            , direction="ltr"
                            , halign="left"
                            , valign="bottom");
```
I get `Too many unnamed arguments supplied` and `Unable to convert mirror(1) parameter to a vec3 or vec2 of numbers` on the `mirror(1,1,0)` line.

mrWheel, 2022-03-04:
From the top of my hat:

The syntax for mirror() is incorrect (mirror([x,y,z]) ??)

udif, 2022-03-04:
I had to define `pcbLength` and `pcbWidth` because they are used by `holeX`, `holeY1`, `holeY2`.  

I also finally realized why all this mess is happening:
`holeX` etc. is used by `pcbStands`. since `pcbStands` is used inside `YAPPgenerator_v13`, all the constants have to be defined before the library include.

The `mirror()` command was not modified by me, it's your original code, but something in the changes is causing this warning.
The strangest is the warning on `getMinRad` since it is a fully contained function, and shouldn't have been affected by any outside change.

I will comment on the extension next to your comment. I didn't specify it yet, because it is too early for a pull request.

udif, 2022-03-04:
Upon further checks, the warnings on `YAPPgenerator_13.scad` (not the warning I discussed on my Colorlight code) appear on your code too:
If you open `YAPP_ArduinoUno.scad` from your `main` branch, you will see the same errors.

mrWheel, 2022-03-04:
> Upon further checks, the warnings on `YAPPgenerator_13.scad` (not the warning I discussed on my Colorlight code) appear on your code too: If you open `YAPP_ArduinoUno.scad` from your `main` branch, you will see the same errors.

Well .. I don't. What version of openScad are you using?

I'm using **version 2015.03-3** witch might be the problem ...

I will upgrade to the latest and see what happens ...

mrWheel, 2022-03-06:
Ok, I understand.

Great work by the way!

First: I think you should drop the offset parms.
This is far to complicated for an end user who will probably just measure the position of the cutouts. I suggest you make the center of rotation the normal xy of the rectangle or, if set, the center of the rectangle.
You can than just add a last parm (angle) that is used if it’s there or not used if not.
As it is now it is too much targeted at a specific goal.

i’m thinking of something like this:

<pre>
//-- base plane    -- origin is pcb[0,0,0]
// (0) = posx
// (1) = posy
// (2) = width
// (3) = length
// (4) = angle
// (5) = { yappRectangle | yappCircle }
// (6) = { yappCenter }
cutoutsBase =   [
                 [30, 0, 10, 24, 35, yappRectangle]                       // angle = 35 degrees
               , [pcbLength/2, pcbWidth/2, 12, 4, 0, yappCircle]          // circle does not have an angle
               , [pcbLength-8, 25, 10, 14, 0, yappRectangle, yappCenter]  // angle = zero degreed
                ];
</pre>

Second: Does this work for cutouts on all planes?
If not, please append and test!
Preferable create a “demo” box with angled rectangles with different angles and sizes on all six planes.

Third: Also append the “documentation” in front of the cutOut functions.

Last: remove all not-used array rows from you example.

udif, 2022-03-07:
Your suggestion to drop the rotation center does not work for me, as it forces me to do some non intuitive calculations for the new cutout offset.

In any case, I'm too unproductive in OpenSCAD as a platform, and frankly I gave up on it. I never liked the fact that I need to learn a dedicated programming language just for 3D objects, and I don't like it's declarative nature either.  
I prefer to use a standard programming language (JS, Python) with 3D bolted on top, such a [OpenJSCAD](https://github.com/jscad/OpenJSCAD.org/) or [CascadeStudio](https://github.com/zalo/CascadeStudio) .  
For JS this has the added bonus of being able to edit or share my design on the web without requiring a local client.

So at the moment I'm converting your code to run on [OpenJSCAD](https://github.com/jscad/OpenJSCAD.org/).
I'm ~25% into the conversion. (OpenJSCAD has an OpenSCAD converter, but it's old and only conforms the old 2011 syntax).
I will start a new repo when I have something that runs.

Anyway, during the translation I found a few issues you might like to fix on your code:

`function getMinRad(p1, wall) = (p1<=wall) ? 1 : p1-wall;`
Probably does not work for the case `0 < p1 - wall < 1` as it yields a value below 1.
You probably want something like:
`function getMinRad(p1, wall) = (p1<=wall+1) ? 1 : p1-wall;`

In `printBaseMounts()` you have several `} //rotate-Y` , `} //rotate-Z` comments which are not nested properly (purely cosmetic).
Others are just aesthetics (use of `-x` instead of `x*-1` in some places, dropping the use of nested `rotate` on the same axis).
At a later stage I intend to use the more expressive nature of JavaScript to improve the interface and make it more user friendly - e.g. Use of named properties instead of unnamed arrays.
If I'll find more issues while I work on my code, I'll post here.

mrWheel, 2022-03-07:
Ok!
I do understand. I’m not very comfortable with openSCAD myself but had never heard of openJSCAD.
Looking forward to your development!

udif, 2022-03-07:
Feel free to visit their [live playground](https://www.openjscad.xyz/), and look at the demos and online manual.

udif, 2022-03-13:
Just to give a better example of what I'm aiming at:

https://github.com/udif/openjscad-objects-V2/tree/master/YAPP_jscad ,  and more specifically:

[Colorlight 5A-75B](https://openjscad.xyz/?uri=https://raw.githubusercontent.com/udif/openjscad-objects-V2/master/YAPP_jscad/colorlight_5a_75B.jscad)

It is now in reasonable shape, but hardly tested beyond my board.

bdlightner, 2022-03-14:
I'm confused.  Did the user interface just change?

I just gave your "YAPPgenerator_v14.scad" a try and **_all_** of my rectangular cutouts disappeared!  I am using OpenSCAD version 2021.01.

Before (with YAPPgenerator_v13.scad):
[image]

After (with new YAPPgenerator_v14.scad):
[image]

Here is my project file if you want to see this yourself:
http://www.lightner.net/Gen2.5_EMU_Debug_Adapter.scad

This is the PCB:
[image]

This is one of the enclosures:
[image]

BTW: I really find your library **_very_** useful.  I've made three different tiny enclosures for what I'm currently prototyping.

mrWheel, 2022-03-14:
@bdlightner 
Yes, as you can read in the README.md and in the [documentation](https://willem.aandewiel.nl/index.php/2022/01/02/yet-another-parametric-projectbox-generator/) there now is an extra param “angle”.

> All examples are made with version 1.3 of the YAPPgenerator. From version 1.4 on the rows in the cutouts arrays have an extra parameter “angle”. For existing designs it suffices to add them with the value “0” (zero) directly before yappRectangle or yappCircle.

udif, 2022-03-14:
I don't understand why you have to break compatibility for your users. You could keep the angle as a last parameter for compatibility, and use the `len()` function to see if it is present or not (defaulting it to 0)

mrWheel, 2022-03-14:
@udif 
It is just more in line with the whole array-row concept. All the “enumerated” var’s are at the end of the row for all array’s.

closed 2022-03-14 by mrWheel

udif, 2022-03-14:
It's a pity you had to break compatibility.
I think If you already decided to break compatibility, I would have moved the 2 enums to the front, as they determine the meaning of all other parameters. This way new parameters will always add to the end and not break compatibility.

mrWheel, 2022-03-14:
@udif 

As there is a variable number of enum's in a cutout-row it would be hard to determine if (f.i.) the second enum-parameter is `yappCenter` or just the value "1" ...

mrWheel, 2022-03-14:
> I do not see an "angle" parameter for "cutoutsBase" in "library/YAPPgenerator_v14.scad". Was there is reason for that?

My mistake.

It is there, you can (must) use it!