mrWheel/YAPP_Box #7: Enhancement request: threaded inserts support

LightTemplar, 2022-04-19:
I would like to have those things as on the image, which can be used for threaded inserts or just as screw hole. On the lid and/or on the base. 
I want to mount box on a bicycle, so I need some reinforcement.
[image]

mrWheel, 2022-04-20:
General questions and enhancements should be posted as a comment on my website at [this](https://willem.aandewiel.nl/index.php/2022/01/02/yet-another-parametric-projectbox-generator/) post.

closed 2022-04-20 by mrWheel

LightTemplar, 2022-04-20:
Those words were here, on GitHub without referencing your Website. If you set your rules, you have to announce them in a place, where they apply. This is a kind of respect to people. Telling them about some rules only then, when they act - it's a clear disrespect.
And if you deal with people without respect, don't expect the respect from them. 
You can do my request or not, it doesn't matter for me anymore.  
[image]