mrWheel/YAPP_Box #11: Thank you

pjaegerhh, 2022-12-29:
No Issue but a ***big*** public THANK YOU Willem for this project. It's super helpful, saves tremendous amount of time and works like a charm :)

wjcarpenter, 2022-12-31:
Seconded.

AustinHaigh, 2023-02-05:
I agree, great work on this library. It's easy to learn how to use it and it has saved me a huge amount of time.

mrWheel, 2023-11-28:
@faludi,

I will release a new version in about a week.
Much new functions! Most important ‘filets’ on almost al objects.

dtorres-sf, 2024-02-22:
Agree - this is extremely helpful. I focus on electronics and am new to CAD / 3D Printing. This project has allowed me to very quickly and easily build enclosures for my projects! 

Do you have any other way to donate outside of paypal by chance? I would love to donate via credit card, not sure how easy it is to enable github sponsors but I have donated to other projects that way. Unless there is a way for me to donate via paypal without a paypal account, that would work too.

mrWheel, 2024-02-22:
@dtorres-sf ,
As far as I you don’t need a PayPal account to donate.
You can use the [[Buy me a coffee](https://willem.aandewiel.nl/index.php/2022/01/02/yet-another-parametric-projectbox-generator/)] button on my website.

dtorres-sf, 2024-02-22:
@mrWheel You're right, that worked! For some reason the link on your github readme forced me to login first. But I was able to do a card donation with your website link. Thanks again for sharing your work!

niccit, 2024-07-21:
Agreed! This library is so helpful. Thank you!

Daroch, 2024-12-25:
I've been thinking a few times about how to make a simple parametric box for my PCB circuits, but your work is awesome! Thanks!!