mrWheel/YAPP_Box #12: are standoffHeight calculations correct for pin hole lengths?

wjcarpenter, 2022-12-31:
In v1.6, the default for standoffHeight is 3.0 mm. For a board I've got, I needed to change that to 4.0 mm. I found that the case does not fit together snugly after that. The pin hole legs are touching the PCB surface, but there is still a gap of (eyeballing) about 1 mm before the snap-fit tabs can engage. Without the PCB inside, the case does close perfectly. It also closed perfectly in an earlier iteration that used the default standoffHeight. 

I'm doing some experiments now to see if faking the PCB thickness by an extra 1 mm will work around this.

I wanted to track this down and send you a PR, but the calculations look a bit complicated. I thought maybe you could find the problem much faster than I could (or would be able to confidently tell me that I'm wrong).

mrWheel, 2023-01-01:
@wjcarpenter ,

can you please provide some code that shows this problem?

wjcarpenter, 2023-01-01:
Perhaps I misinterpreted there being a default for standffHeight. I was looking at this in YAPPgenerator_v16.scad:
```
//-- How much the PCB needs to be raised from the base
//-- to leave room for solderings and whatnot
standoffHeight      = 3.0;
```

Sure. Here's the box I'm currently working with. https://gitlab.com/wjcarpenter/ventbot/-/blob/c5b6ebb2ac1357e48ecf20c9c1c287f135b30327/ventbot_yapp.scad (I've pointed to a specific commit so you won't be distracted by any updates I make later.)

It's a bit complex, but I thought it better to quickly give it to you as-is instead of waiting to simplify it. Obviously, you can just remove or ignore all the stuff for cutouts and labels. Let me know if you need me to simplify it.

You can see at line 57 that I added 1mm to the PCB thickness. (`ki_pcb_thickness = 1.6 + 1.0;`). I verified with a print that that works around whatever the issue is. Without that compensation, I see the problem I originally described.

wjcarpenter, 2023-01-01:
I had some time this morning, so here is a simplified version of that same file. My 3D printer is busy for the next few hours, but I'll eventually do some prints to see if my simplified version has the same issue.

[ventbot_yapp.scad.txt](https://github.com/mrWheel/YAPP_Box/files/10329267/ventbot_yapp.scad.txt)

mrWheel, 2023-01-01:
I tested this morning with the YAPP_testbox.scad and did not find any problemen with any standoffHeight or pcbThickness or combination of these.

wjcarpenter, 2023-01-01:
Thanks for looking into it. I'll poke around some more and see if I can narrow it down. It'll be OK with me if I find out I've done something wrong. :-)

wjcarpenter, 2023-01-01:
I've been using a copy of the repository that I cloned in early December, but I see there have been a couple of commits to the 1.6 generator since then. I'll see if that makes a difference.

bdlightner, 2023-01-01:
> You can see at line 57 and 207 that I added 1mm to the PCB thickness. (`ki_pcb_thickness = 1.6 + 1.0;`, `pcbThickness = ki_pcb_thickness;`). I verified with a print that that works around whatever the issue is. Without that compensation, I see the problem I originally described.

**wjcarpenter** Please share what 3D printer model, nozzle size and layer height you are using for your enclosure prints, as well a the filament type and manufacturer.  I routinely add ~1 mm to the PCB thickness for manufacturing "tolerances".  That applies to high volume injection molded plastic case designs _**and**_ when using **mrWheel's** excellent parametric enclosure 3D model generator.  IMHO we need to rule out the actual quality of your 3D prints before you make **mrWheel** jump through too many more hoops!! :-)

wjcarpenter, 2023-01-01:
Well, yeah, I get that, for sure. It was amazing to me when I printed my first copy of this that it fit together so well. Impressive. 

The thing is, I'm printing both "good" and "bad" with exactly the same printer, slicer settings, filament, etc. I can provide all those details later (it's not some big secret), but I kind of doubt that's what's behind it.

bdlightner, 2023-01-01:
**wjcarpenter**: Not to be too snarky about this but your printer settings matter!  Please share them.

For example, if you are using an 0.3mm slicer height you cannot expect tolerances to be better than ±0.3 mm.  The 3D printed enclosure "faces" securing the top and bottom of your PCB end up being "rounded" to the nearest multiple of your selected layer height.  A tiny change to the 3D model can easily change the beginning and end Z-axis "gap" printed to hold your PCB.  One print fits and a successive one does not likely because this "gap" starts and ends at different Z heights because of an STL model change.  Look at the "stepped" rails and roof of the widely used Bench boat 3D printing standard.  Surely you've printed that?

wjcarpenter, 2023-01-02:
I've been doing a lot of experimenting today. Instead of printing the enclosures, I'm taking measurements in my slicer. (Using the slicer UI to trim away the sides of the enclosure for both base and lid, then precisely snipping off the base pin that sticks through the PCB mount hole, leaving only the stub on which the PCB rests.) No matter what things I have tried, the sum of the Z dimensions for the base and lid always works out to be exactly correct. I tried these experiments with both the snapshot YAPPgenerator that I took in early December and the current HEAD of the master branch.

So, I think there probably _IS NOT_ a problem in the YAPPgenerator code. The print that came out "wrong" is maybe "just one of those things", as @bdlightner has been suggesting. I'm going to keep the extra 1mm of fake PCB thickness anyhow since those prints are coming out great, and for this project it's not too important that the PCB be clamped tightly (at least within that small amount).

FWIW, all of my prints have been done with PrusaSlicer at 0.16mm layer height on a Creality Ender 3v2 using JAYO PLA+ filament.

Thanks to those who followed along on this issue. I'm closing it out now.

mrWheel, 2023-01-02:
@bdlightner,

thanks for enlightening us with your insight knowledge of 3D slicers and printers!