mrWheel/YAPP_Box #20: feature request: through hole in base, through standoffs, into lid standoffs

wfdudley, 2023-05-30:
Your script is incredible.

I would like to fasten my boxes together by putting a machine screw in from the base, through a board standoff, through the board, and threaded into the matching lid standoff (which can already be programmed to have a hole in it.)

I don't mind threading the the lid standoff with a tap (or heat-setting a metal threaded bushing), so no threads need be programmed. 

The base should have some sort of recess for the screw head.  A simple cone would be a good start, though that might be problematic for pan head screws  (but fixable with some hand work).

I have successfully created both lid and base standoffs with holes in them, but the base hole does not pass all the way through the base.  Of course, I can fix this by hand with a drill.

I've tried hacking away at v20 of your library, but I'm not good enough at openscad to get the hole to pass all the way through
the base.

Thanks again for a great script.

mrWheel, 2023-05-31:
Hi @wfdudley,

Please place this comment on my webside.

closed 2023-05-31 by mrWheel

wfdudley, 2023-05-31:
Sorry, I should have read the docs.

Bill Dudley

This email is free of malware because I run Linux.