mrWheel/YAPP_Box #21: Add option for standard DIN Rail mount

studiobts, 2023-09-14:
Hi,
there are many cases where PCB will be put into rack or electric cabinet, so will be very useful to have an option to enable/disable a standard DIN Rain mount on the base.

Any possibility to add in a future release?

Many thanks.

mrWheel, 2023-09-14:
@studiobts 
It certainly would be a nice future but one you can already add your self with the `baseHookOutside` and some programming.
I will not add a specific function for this in the near future.

closed 2023-09-14 by mrWheel

studiobts, 2023-09-15:
Okay, thank you anyway for the fantastic project.

 I'm totally new to OpenSCAD, I will try to create something and if I will succeed I will share the code so you can evaluate it to integrate in a future release.

Thank you.

mrWheel, 2023-09-15:
That is the best way to learn!
Start with making a rail mount without the YAPP-generator.
If you succeed you can try to integrate it with the YAPP-generator. If neededI will try to help.