mrWheel/YAPP_Box #54: Request: re-organize the repo so that version are tracked at historical commits, and are available as releases

DeflateAwning, 2024-01-05:
The proper way to use a Git repository is to only have a single version of the project in the repo at one time. Versions should be tracked by tagging them using GitHub Releases. 

If you want to continue to support historical versions (e.g., you have v19.0, but you want to release a small improvement called v19.1), you should use branches for each of those lineages. This is how projects like OSGeo's projects (e.g., QGIS) work, and is rather standard.

Let me know if you'd like any more guidance getting this setup!

mrWheel, 2024-01-05:
Thanks for your message.

I hardly understand how github works and I’m glad I know how to commit and push updates to github.

I work locally on my iMac where I clone a version from the current state from github and after changes and testing I commit and push from my iMac to github.

If there is a better way I like to learn.

Regards,

Willem 
The Netherlands 
gsm: 0653 - 499 551
www: http://willem.aandewiel.nl/
---
Thank God I'm Atheist

Verstuurd vanaf mijn iPad

DeflateAwning, 2024-01-05:
Starting from the ground-level, you'll want to make sure you're using GitHub Desktop. Clone the repo to a folder on your computer, and don't use the online GitHub interface to upload/update files (if that's what you were doing).

Second, responding to GitHub Issues using the online interface works a little bit cleaner than responding by email.

Next, what I would do is make 2 branches off of the `main` branch, using GitHub Desktop: one called `v1`, and one called `v2`. To do this, check out `main`. Click "New Branch", and name it `v1`. Then, check out `main`. Click "New Branch", and name it `v2`.

Checkout the v1 branch (switch to the `v1` branch). Delete all the files that aren't v1-related. Optionally, rename all the files to remove the version numbers from the files (you don't generally include version numbers in filenames with Git). Commit the changes, and push them. Now, in the GitHub Online interface, publish a new release of this v1 branch. In the "tag" section, name it `v1.9.0` (as that's the latest v1 version, it seems).

Now, check out the v2 branch. You'll see that all the files for all version re-appeared. Do the same things as for v1, but for v2. Tag a release; as an attachment to the release, add the `YAPP_Template_v30.scad` file, the `YAPPgenerator_v30.scad`, and the `roundedCubes.scad` (as these three files are the meat-and-potatos of your whole project). You can ZIP them, or upload them directly. 

Now, check out the `main` branch. From now on, this `main` branch will contain only the latest version of the code. Delete all v1- and v2-related files, and commit that. Optionally (but recommended), remove `_v30` from all the filenames. Next time you do a release, you'll do it from the main branch, and you'll tag it as v3.1.

If you'd like me to do all this, feel free to add me as a Contributor to this project and I can help out. Or let me know if you have any questions.

mrWheel, 2024-01-07:
Ok, your offer is tempting.

But I still have questions…

Why do I need to install the github desktop version?

Say the main branch is, after your work, v3.1.0
What are the steps to prepare the repo for some new code or bug fixes? It is just not clear to me.
Do I have to first make a new tag for say v3.1.1 and than do something to make that v3.1.1 “active”?
Say the changes did not workout, can I than remove v3.1.1?

DeflateAwning, 2024-01-08:
You can use the command line version of `git` if you want, but GitHub Desktop is a little easier to get started with, and has a way easier way to quickly preview what files have changed since the last commit.

You create tags, which are effectively pointers to the codebase at a specific commit.

If you just tagged and released v3.1.0 in the main branch, and you want to start working on v3.1.1, you just start making edits/commits in the main branch. When you're done, commit your changes, and tag a release from the main branch.

You only commit when you're happy with your changes, in general. You can revert commits that you're not happy with.

You can (and should) have multiple commits between tags.

mrWheel, 2024-01-08:
@DeflateAwning 

> You can use the command line version of git if you want, but GitHub Desktop is a little easier to get started with, and has a way easier way to quickly preview what files have changed since the last commit.

What I ment is; “why the desktop instead of the web interface”.

The nice thing about the web interface is that it also works on my iPad. But if you really think it has benefits I will install the desktop version.

DeflateAwning, 2024-01-08:
Have advantages yeah. The bigest one is that the files are stored on your computer, and you don't have to worry about diverging files.

Git is not Google Drive. Trying to use Git as a Google Drive alternative means you're missing out on the main features and workflows of Git. I'd recommend following a tutorial through to get a hang of how it's intended to be used.

rosenhauer, 2024-01-08:
I'm not familiar with Git either (come from VSS source control at work) but I agree that YAPP should be setup however is best/proper for Git.  I can learn with you on how to do it properly.

mrWheel, 2024-01-11:
@DeflateAwning ,

Added you as a collaborator.
Please do your magic!

One remark though: The reason for keeping files with version in there name is for users who have made projectboxes with that (old) version of the Yappgenerator. I would like them to be able to keep using those projectbox files…

But I’m fine with having the main branch without a version number and maybe those having “old” files should organize this themself..

DeflateAwning, 2024-01-11:
Awesome, will make sure they're still around! Thanks a lot for adding me; I'll take a look at this in the next couple days.

DeflateAwning, 2024-01-15:
Alright, all the changes are made! There are now three primary branches going forward:
* `v1`, which tracks all files going forward in v1 (which may be unnecessary, as I doubt anyone is still making upgrades to v1)
* `v2`, which tracks all files going forward in v2, in case there are any more v2 upgradesr
* `main`, which currently contains only the latest version of the v3 code.

In the README, I added a versioning section at the bottom. That's how best I think the code in this repo should be handled, where a new major release is created anytime a breaking change is made to the API. 

In [my final commit](https://github.com/mrWheel/YAPP_Box/commit/bb8156f655213246126b88c9112d6616bfb08bc6), I set the "version" to `v3.0.1`. It'd be awesome if you could tag a GitHub Release today and upload the two main files (`YAPP_Template_v3.scad` and `YAPPgenerator_v3.scad`)! 

Let me know what you think! I'm happy to help make any changes/reversions if you spot any issues!

rosenhauer, 2024-01-15:
Can we do basic workflow on the issues in Git. That is have them approved and assigned or at least marked as to who is working on them?  

I only see close as fixed and close

rosenhauer, 2024-01-15:
Didn't mean to close it. Was just trying to look at the options and clicked the wrong spot.

mrWheel, 2024-01-15:
@DeflateAwning ,
You did a great job organizing this repo. Even updated the examples to the new location of the YAPPgenerator!

Bud .. by changing the directory structure you can no longer just clone or download the repo and get working on your own files (I liked the `library` map with the YAPPgenerator and the roundedCubes programs in it).

I suppose I have to change the documentation for that.

Don’t know what you mean by 
> it’d be awesome if you could tag a GitHub Release today and upload the two main files (YAPP_Template_v3.scad and YAPPgenerator_v3.scad)!

.. normaly if I create a Release all the files are placed in it. What tag do I give this release?

DeflateAwning, 2024-01-15:
Thanks! 

> by changing the directory structure you can no longer just clone or download the repo and get working on your own files

What do you mean? You just have to make sure that the path to the library is correct. I'd suggest that the best practice is to always put a copy of the YAPPgenerator file alongside the box so that you have a copy of it at the right version. 

I removed the `roundedCubes.scad` dependency because it's not used in v3. I'd suggest that BOSL2 is a better way to generate round cubes nowadays. 

Some minor docs changes may be required. But happy to discuss reverting anything you're not happy with.

When you create a release, you can attach additional files so that they're available to download directly (instead of having to go through the process of downloading the ZIP file). I would give this release the tag v3.0.1, as it seems like a v3.0 has already been tagged. v3.0.1 is what's in the SCAD files, also.

Very happy to help explain anything! This is a fantastic library, and I'm happy to help make sure it stays maintained as long as possible!

mrWheel, 2024-01-15:
I mean that I have a directory structure as in the documentation:

```
YAPP_box
      \— library
              \— all versions of the YAPPgenerator
   — all my boxes
```
I can still create this but for the novice user it is a bit of a hassle to set this right.
But I can and will change the documentation on this.

> Some minor docs changes may be required. But happy to discuss reverting anything you're not happy with.

No need for that. I recognize the structure you now created from many other repo’s.

> When you create a release, you can attach additional files so that they're available to download directly (instead of having to go through the process of downloading the ZIP file). I would give this [...]

Ok, created the release with tag v3.0.1. Doing it from the sofa (with my iPad and the github web interface) which did not give me the option to add the two .scad files. When I’m in my workshop I will add them.

> I removed the roundedCubes.scad dependency because it's not used in v3. I'd suggest that BOSL2 is a better way to generate round cubes nowadays.

Looking at the BOSL2 library documentation it seems there is a special place for global libraries for OpenSCAD.

- windows: My Documents\openSCAD\libraries
- Linux: $HOME/.local/share/OpenSCAD/libraries
- MacOS: $HOME/Documents/OpenSCAD/libraries

Maybe I should focus on that and suggest in the documentation to place the YAPPgenerator_vx.scad there…

I am always amazed at the help that complete strangers provide via the internet.
Thanks you for your help and patience.

mrWheel, 2024-01-15:
> I was actually going to suggest that the global library option may be good for this project; the downside is that it removes portability between computers, if you use a tool like Google Drive to sync your files to multiple computers. Might be worth mentioning both options so that people can pick the one that works best for them.

I have no experience with Google Drive so I cannot test or say something useful about Google Drive but I will mention the global library option in the documentation.

> It's easy to "edit the release" and add the files whenever you get a chance. Normally you'd attach the compiled executables if it was a regular software project, but for this, it makes sense to attach the SCAD files there.

Done that!

closed 2024-01-15 by mrWheel