mrWheel/YAPP_Box #56: Request: in `pcbStands`, add option to have pins on the top instead of the bottom

DeflateAwning, 2024-01-08:
Currently, you configure which sides (base/lid) have pins and holes with these options:
```
//    n(a) = { <yappBoth> | yappLidOnly | yappBaseOnly }
//    n(b) = { <yappPin>, yappHole } // Baseplate support treatment
```

`yappBoth` puts the pins on the bottom, and the regular connectors on the top.

## Option 1 (preferred):
I'm proposing that the section could instead be setup like this:
//    n(a) = { <yappBasePin> | yappBaseHole | yappBaseNoStand } // bottom/base support config
//    n(b) = { <yappLidHole> | yappLidPin | yappLidNoStand } // Baseplate support treatment // top support config

This option allows easier configuration of the shape of the PCB stands on each side. The downside is that it's possible to configure it so that there are pins on both sides (which would prevent it from closing). It's easy to avoid doing that though.

## Option 2:
Alternatively, I'm proposing that the line could be changed to:

```
//    n(a) = { <yappBothPinsFromBottom> | yappBothPinsFromTop | yappLidOnly | yappBaseOnly }
//    n(b) would stay the same
```

This change would prevent the user from creating pins on both the top and the bottom, but it feels way less intuitive to me.

rosenhauer, 2024-01-15:
I think option one makes more sense.  It can be trapped to raise a message if a pin is selected on both.

lijon, 2024-01-15:
I just came here to ask for the same feature :)

rosenhauer, 2024-01-26:
Fixed in #62

closed 2024-01-26 by rosenhauer