mrWheel/YAPP_Box #57: Change pin height

mrWheel, 2024-01-15:
[Issue](https://willem.aandewiel.nl/index.php/2022/01/02/yet-another-parametric-projectbox-generator/#comment-4760)

[Jonatan Liljedahl](http://kymatica.com/) says:

Is there any way to change the pin height? I want to mount some parts almost flush to the lid, for example a display. But changing the standoff height makes the pin go through the lid.

[image]

rosenhauer, 2024-01-15:
I can look at adding that with the other change.  I also have a display opening feature 90% complete that I was planning on adding soon.  It allows you to make an opening with a beveled edge and a need pins to attach the display to the lid and snaps to lock it on.  

On my phone now and don't have pictures. But I can share some later.

rosenhauer, 2024-01-15:
[image]

This is what I have so far.  It allows you to adjust the pins and the depth from the actual screen to the PCB of the screen.  

Here is a finished box with a screen. 

[image]
[image]

I'm still working on the 3D printing with the new printer. So I ended up hot melt gluing it in as the snaps didn't fit and I was in a hurry.

mrWheel, 2024-01-16:
Looking good!

rosenhauer, 2024-01-26:
Fixed the Pin Height/length in #62

closed 2024-01-26 by rosenhauer