mrWheel/YAPP_Box #64: YAPP_Box - unexpected behaviour with the padding around the PCB

Dingo61, 2024-01-29:
I am learning how to use settings in YAPP_Templatev3.scad, I have include set to YAPPgenerator_v3.scad (downloaded 27 Jan 2024)

I have the following settings: 
showSideBySide  = true
viewport is set to view from above

I am observing some unexpected behaviour when I change the padding around the PCB
Increasing value of PCB "padding front" increase padding at back and left, but does not increases padding at front ?
Increasing value of PCB "padding back" increases padding at front, but does not increase padding at back ?
Increasing value of PCB "padding right" has correct result.
Increasing value of PCB "padding left" increases padding at right, but does not increase padding at left ?
The unexpected movements of the PCB also causes lid switch cutouts and other related PCB artifacts to move to incorrect positions in the lid

Can you check and see if you can replicate this behaviour? 
Are there other settings I need to enable first ?

I think the behaviour lies in YAPPgenerator_v3.scad, 
function getPCB_X(pcbName="Main")
 lines 4497, 4498 and 4502 all reference 'paddingFront', is this correct?
should line 4497 be 'paddingBack' ?
should line 4498 be 'paddingLeft' ?
should line 4502 be 'paddingRight' ?

Regards

rosenhauer, 2024-01-29:
Good catch.  

I seem to have messed that up when I added multiple PCB support.  

Will get an update ASAP.

rosenhauer, 2024-01-29:
Hopefully I did the merge correctly.  I tested it with the Multiple_PCB_v30.scad example and it now adds the padding in the correct places.

closed 2024-01-29 by rosenhauer