mrWheel/YAPP_Box #72: Feature request: add option to round the vertical edges, but not round the edges touching the print bed

DeflateAwning, 2024-02-26:
I'm never that happy with how the rounded edges in the first few layers turn on, for the edges along a face touching the print bed. It would be awesome if there was an option to only round the vertical edges, and not round the printbed edges.

rosenhauer, 2024-02-26:
I've thought the same thing.  Unfortunately it's not a trivial change, but
something I could still look at in the future.

DeflateAwning, 2024-02-26:
The 45deg idea is a good one, will give that some thought! Easy with Prusa's new organic supports.

It would be cool if this library considered using BOSL2 to generate the rounded box. It can generate the box very easily with its `cuboid` API. The BOSL2 library could be included in this repo as a git submodule. Downside, of course, is that it adds complexity for others using this library; none-the-less, we could maybe copy over just the `cuboid()` function from BOSL2.

goramai, 2024-02-27:
All rounded edges are just fine for my boxes printed by JLC3DP (I don't have a 3D printer). Please make any new behavior optional.

DeflateAwning, 2024-02-27:
Of course any new behaviour will be optional; that was in the feature request (which reads "Feature request: add option...").

PR #109 (2024-09-09): Add new option for boxType.
Add option for changing the base box shape.  The default is the rounded box that we all know. But there are now options to have square edges/chamfered edges and the bottom edges different than the vertical edges.

Samples of the types are below:

[image]
0 - The Default rounded corners/edges. 
[image]
1 - All edges square
[image]
2 - All edges chamfered
[image]
3 - Square top and bottom edges (the ones that touch the build plate) and rounded vertical edges
[image]
4 - Square top and bottom edges (the ones that touch the build plate) and chamfered vertical edges
[ima [... cut]
closed 2024-09-09 by mrWheel