mrWheel/YAPP_Box #76: Clarify include and function needed for basic operation

zyphlar, 2024-03-12:
Following what's in Git and GitBook it's not mentioned that you need `include <YAPP_Box-3.0.1/YAPPgenerator_v3.scad>` at the top of the file and `YAPPgenerate();` at the bottom. Only by reading the blog post did I infer those details.

Ideally in the What You Need section of the Book and the How to program your Project Box section of the Readme these details will be added so that a newbie can copy-paste and get basic output in OpenSCAD before continuing, sort of a Hello World to ensure that things are installed correctly.

Indeed the instructions talk about putting YAPP in the libraries folder, but then the blog post tells you to include the file with a relative path, which can confuse people who aren't experts.

Thanks for this great library!

mrWheel, 2024-03-15:
@zyphlar,

You are right!
But I’m currently very busy with an other project and don’t have the time for at least the next month!

mrWheel, 2024-06-11:
I have updated "[What you need](https://mrwheel-docs.gitbook.io/yappgenerator_en/what-you-need)" and "[Getting Started](https://mrwheel-docs.gitbook.io/yappgenerator_en/getting-started)" in gitbook.

Please read and give comment's.

closed 2024-06-11 by mrWheel

zyphlar, 2024-06-12:
Looks good! The only part I noticed was that "It is wise to leave the folder structure as it is and save new box designs in this “YAPP_Box-main” folder." is obsolete with all the other instructions about the pros and cons of where to install the library

Thank you!

mrWheel, 2024-06-12:
> The only part I noticed was that "It is wise to leave the folder structure as it is and save new box designs in this “YAPP_Box-main” folder." is obsolete with all the other instructions about the pros and cons of where to install the library

I made some minor changes to the text.

zyphlar, 2024-06-12:
Looks good thanks!