mrWheel/YAPP_Box #81: showPCB = true  but PCB not at the right height!

mrWheel, 2024-03-22:
I'm currently working on a box for a project I'm working on.

YAPPgenerator: Version="v3.0.3 (2024-03-21)";
But with showPCB = true the PCB is not displayed at the correct height.

- showPCB = true;
- standoffHeight = 4.0;
- connectors   = [  [5, 5, 4, 2.5, 6 + 1.25, 4.1, 9, yappDefault, yappAllCorners, yappCoordPCB]  ];

[P1Reader2Eth_scad.txt](https://github.com/mrWheel/YAPP_Box/files/14719134/P1Reader2Eth_scad.txt)

As a consequence the switch-tubes are also incorrect (to long)!

[image]

# Also - future request:
I would like the Cap of the push-buttons to be a bit higher so they fall a bit deeper in the lid. With tolerances during printing sometimes the Cap is completely above the lid. No problem for a yappCircle cap but if it had a (f.i.) yappRectangle shape it can turn and then you can not push it anymore. 

[image]

Also a pole/hole slack variable would compensate for printer and material tolerances.

rosenhauer, 2024-03-23:
There was an issue with the PCB thickness being removed from the Connector height and causing it to be off. 

But an additional issue is that you used an older definition for the Connectors:
Below is the update to the newest version.

Few changes to note: p(2) = StandHeight is from the coordinate system origin so in the case where yappCoordPCB is specified (new default) it should be set as 0 unless you need to to be different than the PCB position.
```

//===================================================================
//  *** Connectors ***
//  Standoffs with hole through base and socket in lid for screw type connections.
//-------------------------------------------------------------------
//  Default origin = yappCoordBox: box[0,0,0]
//  
//  Parameters:
//   Required:
//    p(0) = posx
//    p(1) = posy
//    p(2) = StandHeight : From specified origin  
//    p(3) = screwDiameter
//    p(4) = screwHeadDiameter (don't forget to add extra for the fillet)
//    p(5) = insertDiameter
//    p(6) = outsideDiameter
//   Optional:
//    p(7) = insert Depth : default to entire connector
//    p(8) = PCB Gap : Default if yappCoordPCB then pcbThickness else 0
//    p(9) = filletRadius : Default = 0/Auto(0 = auto size)
//    n(a) = { yappAllCorners, yappFrontLeft | <yappBackLeft> | yappFrontRight | yappBackRight }
//    n(b) = { <yappCoordPCB> | yappCoordBox | yappCoordBoxInside }
//    n(c) = { yappNoFillet }
//    n(d) = { yappCountersink }
//    n(e) = [yappPCBName, "XXX"] : Specify a PCB. Defaults to [yappPCBName, "Main"]
//-------------------------------------------------------------------
connectors   =
[//  0, 1, 2, 3,     4,      5,   6, 7, ...
  //[5, 5, 4, 2.5, 6 + 1.25, 4.1, 9, yappDefault, yappFrontLeft, yappFrontRight, yappBackLeft, yappBackRight, yappCoordPCB]
//    [5, 5, 4, 2.5, 6 + 1.25, 4.1, 9, yappDefault, yappAllCorners, yappCoordPCB]
    [5, 5, 0, 2.5, 6 + 1.25, 4.1, 9, yappDefault, yappDefault, yappAllCorners]
];

```
All of the other dimensions seen to be correct to me.  The PCB and tube length appeared wrong before because the connectors were wrong.

rosenhauer, 2024-03-23:
On the button pocket depth is the same as the lid thickness.  No matter is the button is to be flush or above the lid.   I'm not sure how there could be that much printing error. 
[image]

Could this possibly be related to the above issue where the connectors were off by PCB Thickness.

mrWheel, 2024-03-23:
@rosenhauer,

I started from the YAPP_template_v3.scad that is supposed to be up-to-date with the YAPPgenerator_v3.scad.

Thanks for all your great work!

closed 2024-03-23 by mrWheel