mrWheel/YAPP_Box #95: Reusable cutouts

vladoATL, 2024-04-19:
I'm looking for a solution to reuse my custom cutout. I need this to position the holes for the electronic connectors that are bolted on. This code generates such a cutout, however I don't know how to place it in the template.

//  Parameters:
//   Required:
//    p(0) = connector opening diameter
//    p(1) = horizontal distance between two screw openings
//    p(3) = wall thickness 
//   Optional:
//    p(4) =  count of screw openings arround connector <4>
module connector_w_screws( hole_dm, screw_dist, screw_dm, wall_thickness , screws_cnt = 4) {
    wall_thickness = wall_thickness + .01;
	translate([-(2*screw_dm + screw_dist)/2,-(2*screw_dm + screw_dist)/2,-wall_thickness  / 2 ])
    {
        difference() {
            plate_l = screw_dm + screw_dist ;
            cube( [plate_l,  plate_l, wall_thickness  ]);   // base plate   
            
            translate( [plate_l/2, plate_l/2, -0.01] )                 
            cylinder( d = hole_dm, h = wall_thickness  + 0.02); // connector opening
           
            trans_screw = [
                [screw_dm/2, screw_dm/2, -.01], 
                [screw_dm/2 + screw_dist, screw_dm/2 + screw_dist, -.01], 
                [screw_dm/2 + screw_dist, screw_dm/2, -0.01], 
                [screw_dm/2, screw_dm/2 + screw_dist, -0.01]
            ];
            // screws openings:
            for (i=[0:screws_cnt-1])
            {
                translate(trans_screw[i] )
                cylinder( d = screw_dm, h = wall_thickness  + .02);
            }         
        }
    }
}
connector_w_screws( 24, 26, 4, 1.8);

rosenhauer, 2024-04-19:
You could call that from one of the hook functions.  You need to rotate and translate as needed to get it to the correct position.  

Our if you aren't worried about it extending both inside and outside the case you can just add the call after the generate command at the end of the template.

vladoATL, 2024-04-21:
OK, I tried to add it to a hook function but it wotks only with Lid and Base. I cannot attache it to sides where the connectors are usually placed.