mrWheel/YAPP_Box #96: Generation fails and I don't understand why

XavierP56, 2024-06-02:
Hi, 

The following code gets an error when I render it:
ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron. 
What do I do wrong ?
Thanks.

```
include <../YAPPgenerator_v3.scad>

printBaseShell        = true;
printLidShell         = true;
printSwitchExtenders  = true;
showSideBySide=true;

baseWallHeight      = 25; 
lidWallHeight       = 10;

paddingBack        = 40;

pcbLength           = 91; // Front to back
pcbWidth            = 50; // Side to side
pcbThickness        = 1.5;

standoffHeight = 3;
standoffDiameter = 5;
standoffPinDiameter  = 2;
standoffHoleSlack=0;

pcbStands = [
   [5, 5, yappBaseOnly, yappHole ],
   [5, pcbWidth - 5, yappBaseOnly, yappHole],
   [62, 5,yappBaseOnly, yappHole],
   [62, pcbWidth - 5, 2, yappBaseOnly, yappHole],
];

cutoutsFront =  
[
    [3,pcbThickness,0,0,3,yappCircle, yappCoordPCB],
    [40,pcbThickness,0,0,3,yappCircle, yappCoordPCB],
];

cutoutsLeft =  
[
    [34, pcbThickness, 10, 10, 0, yappRectangle, yappCoordPCB],
    [8, 8, 0, 0, 3, yappCircle, yappCoordBoxInside ],
    [28, 8, 0, 0, 3, yappCircle, yappCoordBoxInside ]
];

areation = 10;

cutoutsBack =  
[
    [15, baseWallHeight - 15, 20, 3, 0, yappRectangle, yappCoordBoxInside],

];

cutoutsLid = [
    [22,42,0,0,4,yappCircle, yappCoordPCB],
];

snapJoins   =   
[
    [10, 5, yappLeft, yappRight, yappSymmetric]
   ,[10, 5, yappFront, yappBack, yappSymmetric]
];

//showPCB = true;

YAPPgenerate();
```

rosenhauer, 2024-06-02:
What version of OpenSCAD are you using?

I was able to generate your code (with only changing the path to the YAPP library, as mine is in an odd place).
[image]

I tested against [OpenSCAD](https://www.openscad.org/) version 2021.01

XavierP56, 2024-06-03:
Same I'm using OpenScad 2021.01 on Windows 11.
Problem occrus when you render (F6) or export STL.
You need to clear cache to see the error. Clear case + F6 ==> error.
Then the generated STL has trouble printing the lid (especially if you remove yappBaseOnly, yappHole in pcbStands)

rosenhauer, 2024-06-03:
A fix has been released for this. Can you please update and confirm the fix.

rosenhauer, 2024-06-03:
Please try again with additional fix.

XavierP56, 2024-06-03:
Looks good. I've generated the STL and able to slice it in Bambu Studio as well.

XavierP56, 2024-06-03:
Closing: I will reopen if I see another issue. Many thanks !

closed 2024-06-03 by XavierP56