mrWheel/YAPP_Box #101: Feature request: screw holes through lid

ettom, 2024-07-02:
Hi,

This is request for the feature described/discussed in https://github.com/mrWheel/YAPP_Box/pull/28. To recap - imagine a wall-mounted box. More often than not, it would be useful to have the back/base (i.e. the side holding the PCB) of the box mounted to the wall while having a removable lid. Currently this only seems to be possible so that to remove the lid, you'd have to first unscrew the base from the wall, as the screw heads are only accessible from the bottom of the base. A solution would be to add a parameter like `yappThroughLid` to the connectors array - if set, the screw holes would be generated through the lid instead. I'd even go as far as to suggest that maybe this could be the default (which would probably mean adding a `yappThroughBase` parameter instead).

rosenhauer, 2024-07-03:
Before I start making changes I want to be clear on what you are requesting.

[image]
Like the second one?  The original is the first one with the arrow.  This was just a quick and dirty image edit - I don't have the code done yet. 

We won't change the default from existing for any new feature so we don't break any existing uses.  But we can add new features.

ettom, 2024-07-03:
Yes, this looks correct.

mrWheel, 2024-07-03:
… there should be a hole on the lid to make it work!

ettom, 2024-07-05:
Looks perfect, thanks!

closed 2024-07-05 by ettom