mrWheel/YAPP_Box #104: Changed file name of .svg to YAPP_Demo_imagesPlane.svg

mrWheel, 2024-07-05:
@rosenhauer ,

Great job, again!

I had, in github, changed the `.svg` file name to the name of the `YAPP_Demo_imagesPlane.scad` program where it is used. 
Just so it is clear the two files belong together.

But that changes seems to be lost. 

I have not checked whether the `.svg` file is used in other Demo programs in which case it is ok to leave the name as it was.

rosenhauer, 2024-07-05:
I don't think that it is used in any other example.  I got a notification that it had changed when I was creating the merge request and thought I told it to ignore it.  Maybe I told it to ignore the change you have not to ignore making the change on my end.

mrWheel, 2024-08-26:
Changed the svg filename again.

closed 2024-08-26 by mrWheel

mrWheel, 2024-08-28:
And the changes are gone again ;-)