mrWheel/YAPP_Box #110: Base PCB support with through-hole and sunken screw head?

snhirsch, 2024-10-01:
I am working on a box for a heavy power supply that must be mounted with machine screws from the bottom.  Maybe I'm missing something, but no matter what I do I cannot get OpenSCAD to generate a complete through hole.  It's creating a "pin" hole, but with a blind bottom.  Tried increasing pin length in the hopes it would project through, but that's not working.  Any suggestions or workarounds?  A countersunk area for the screw head would be icing on the cake, but I could always mount feet as a workaround.

mrWheel, 2024-10-01:
You can just make a round cutout on the places you want the screw holen.
With ‘baseHookOutside’ and/or ‘baseHookInside’ you can re-enforce the screw holes.

closed 2024-10-01 by mrWheel

snhirsch, 2024-10-01:
Yes, but unfortunately I do not understand the syntax for the hook routines.  Although the YAPP template itself is clear I'm far from an expert with OpenSCAD coding.  I think I may have come up with a workaround involving bottom holes that are concentric with the support holes.

Thanks for getting back so quickly!

rosenhauer, 2024-10-01:
Are you using pcbSupports or connectors.  I think connectors might be more
of what you are needing.

snhirsch, 2024-10-01:
The connectors seem to always exist in top + bottom pairs.  I would need them strictly on the bottom and located relative to PCB origin.  My approach of "drilling holes" in larger generated PCB supports seems to have worked out well.  Will need to take a close look in the slicer, but I think this will be fine.  Thanks for coding such a marvelous template!