mrWheel/YAPP_Box #111: Hook for custom cutouts?

jantman, 2024-10-02:
I have existing OpenSCAD models of some complex cutouts for a control panel. I really love this project and would like to use it for my new enclosure, but I can't seem to find a way to provide custom cutouts? As far as I can tell, the hook functions only allow you to add geometry, not to `difference()` it from the box.

I was wondering if there is any existing functionality to let me use an existing module to generate custom cutouts, by differencing it somehow?

rosenhauer, 2024-10-02:
I haven't tried it yet but you should be able to create a cube the
thickness of the wall and the height / width slightly larger than your
model. Then create a rectangular cutout through the case where you want the
custom cutout with the same width/height (or slightly larger to make sure
you don't have any gaps.  Then do the difference between the cube and your
model and attach it.  You can either use the hook and add it to the inside
and outside (the hooks only go half way through the case) or just add it
after the generate command (this will add it across the inside/outside
border).

I hope that makes sense.

Dave

jantman, 2024-10-06:
Thanks Dave. I ended up adding my own hooks in for custom cutouts, it was only a few lines of code. Any interest in a patch or PR?

rosenhauer, 2024-10-06:
Sure send it to me and I'll look at it

jantman, 2024-10-06:
Here's a link to the diff: https://github.com/jantman/machine-access-control/commit/c860e23d8b0bcd43c924b47d14e1e0748aece98f

jantman, 2024-10-07:
Noted about the centering; also this didn't seem to work right for cutouts in the sides. I did some more work last night and needed to tweak this a bit more, but am also running into some physical construction issues with my design. I'll try and make another pass on this and get the centering sorted out.