mrWheel/YAPP_Box #118: Feature Request:  Fit adjustment for Lid/Base

sleander1, 2025-01-19:
Looking for a way to adjust the fit between the lid and the base.  My experience so far has been that there isn't enough clearance between the two, and I have to file or sand parts to get them to fit.  Having an adjustable clearance parameter would be useful!

mrWheel, 2025-01-19:
I have the same question. Specifically with small boxes (35x45mm) the base and lid are difficult put together. A extra parameter could solve this.

rosenhauer, 2025-01-19:
I can look at adding a parameter, but I've found that when I get one that it to right it's always that I forgot to test/adjust the horizontal expansion on my printer.  I'm amazed how much a slight change in that setting can make a big difference in the fit 

This is the tool that I've been using:

https://www.printables.com/model/519880-horizontal-expansion-calibration-tool-v40

mrWheel, 2025-01-20:
@rosenhauer 
Hm… I have a Bambulab X1C and never adjust / calibrate settings so I don’t really know what you’re revering to.

rosenhauer, 2025-01-20:
What slicer do you use?

Cura calls it Horizontal Expansion BamboStudio calls it X-Y hole
compensation.

It took me a long time to figure out that settings importance.

mrWheel, 2025-01-20:
@rosenhauer,

To be honest: I don’t dare to change any setting of BambuStudio as it works perfectly as is is.

What I do know is that in earlyer versions of the YAPPgenerator the Lid cam much more easy over the Base.

So, I still would like a (global) setting to make some clearing between the ridge halves….. but thinking of it: there is a parameter `ridgeSlack`… that probably does what we need!

I will investigate this and then get back to you.

sleander1, 2025-01-20:
I am using Cura with a Ender 3V2 and a sprite extruder.  I bet I could fiddle with flow and layer settings (currently using a 0.6 nozzle), or scale the lid 0.25% larger, but I thought a clearance function might be nice.  Last print I adjusted the layer height which helped on this small box (~ 60x50x20), but it's still pretty tight.  If I add screw connectors, I'd want it to be completely loose.

rosenhauer, 2025-01-21:
If you haven't already downloaded the calibration tool and don't a calibration you should, it makes a world of difference on having parts fit together.  

In still planning on adding a lid slack option, but if you are having to trim/sand the case parts with the default gap then it's a printing issue.  There will already be a gap if the parts are printed accurately.

rosenhauer, 2025-01-21:
> [@rosenhauer](https://github.com/rosenhauer),
> 
> To be honest: I don’t dare to change any setting of BambuStudio as it works perfectly as is is.

From what I've read, and experienced this setting needs to be calibrated with each brand of filament. Some even have found that different colors of the same brand may need to be calibrated differently.  
> 
> What I do know is that in earlyer versions of the YAPPgenerator the Lid came much more easy over the Base.
I don't think the gap has been changed.
> 
> So, I still would like a (global) setting to make some clearing between the ridge halves….. but thinking of it: there is a parameter `ridgeSlack`… that probably does what we need!

I think you are correct that there was some sort of slack setting. I've never messed with it and haven't had time to look at the code yet.
> 
> I will investigate this and then get back to you.

mrWheel, 2025-01-21:
@rosenhauer 

The big advantage of the Bambu ecosystem is that filament is tested in the factory and tagged with RFID, so that the printer has the optimal settings for each roll.

I checked `ridgeSlack` but that setting only moves the half of the ridge forwards or backwards.

I really would like a `lid slack` setting ... no hurry! If you can find the time.

rosenhauer, 2025-01-21:
> [@rosenhauer](https://github.com/rosenhauer)
> 
> The big advantage of the Bambu ecosystem is that filament is tested in the factory and tagged with RFID, so that the printer has the optimal settings for each roll.
That's nice if you use their supplies. I jump brands depending on the best price.

> 
> I checked `ridgeSlack` but that setting only moves the half of the ridge forwards or backwards.
That was how it was working after I made the options of the different case shapes. But it used to be the actual gap.
> 
> I really would like a `lid slack` setting ... no hurry! If you can find the time.
I fixed the ridgeSlack to work again so I didn't add a new setting.  I don't see any reason to move the interface position away from the center.  If you want that I could look at adding it but I can't see why it would be needed (it was just an artifact of the error I introduced before).

#119 Has the Fixes

mrWheel, 2025-01-21:
@rosenhauer,

You are the best!

Thanks

closed 2025-01-21 by mrWheel