mrWheel/YAPP_Box #120: self forming thread

Majrooo, 2025-02-01:
I watched this video:
[https://www.youtube.com/watch?v=HgEEtk85rAY](url)
and I was interested in the part about self-forming thread.
With the help of AI chat I managed to create a cylinder with which such holes can be made. I have modified yappGenerator using the **use** command and replaced the cylinder with a new one, but I am sure others would also welcome such an option via parameters.
The code for the new cylinder is attached.

If possible, please implement this option.

And thank you also for this library, it is very useful.
[self_forming_screw.zip](https://github.com/user-attachments/files/18626677/self_forming_screw.zip)

rosenhauer, 2025-02-01:
Where are you using these? In the connectors/standoffs? 

It looks interesting and I can look at adding this type of screw hole. Willem what do you think?

mrWheel, 2025-02-01:
I think this is a great option to have.

Majrooo, 2025-02-01:
I'm just getting to know the library. So far I've only created the bases and that's why I modified it for use with standoffs. As I look at the manual now, it would also be suitable for connectors. But it can be used anywhere a screw is planned to be used.

And the last thing, do you know how I can download an offline copy of the official documentation for the library?

thanks guys, you're great.

mrWheel, 2025-02-01:
@Majrooo 
Hm.. there used to be a [Create PDF] button on GitBook.

I don't know how to download an offline copy..

rosenhauer, 2025-02-14:
Feature added in v3.3.3 (2025-02-14)

Majrooo, 2025-02-14:
I see a functional implementation for connectors. Is it also planned for standoff?

rosenhauer, 2025-02-14:
Yes, I added it to standoffs in 3.3.4 a few min ago.

Majrooo, 2025-02-22:
Works as expected. Thank you.

closed 2025-02-22 by Majrooo

rosenhauer, 2025-02-22:
Thanks for the suggestion and the sample code. 

I haven't had a chance to try it out myself yet.