mrWheel/YAPP_Box #121: Feature Request - yappCenter for labels

lagomorph, 2025-02-06:
First of all let me say this project is absolutely fantastic. It was just what I was looking for and the design is really good. I can generally get objects exactly where I want very efficiently but with labels I have to adjust position by trial and error due to the size of the text. I think having yappCenter apply to labels would solve this. Perhaps there is another way that I'm missing. Thanks again for this great project.

rosenhauer, 2025-02-14:
There was already a option to center text (both horizontally and vertically) but the documentation in the template was out of date.  The template and generator have been synced.  I also added a Demo on using the options as YAPP_Demo_Labels_v3.scad.

Let me know if this doesn't meet your needs.

lagomorph, 2025-02-16:
This works beautifully, thanks!