mrWheel/YAPP_Box #132: I'm confused. Why do insert-holes not show up in simple design (BUG!?)

mrWheel, 2025-08-20:
@rosenhauer ,

After a long time I need a projectBox so started with the PCB size and the connectors.
But I don't see the holes for the inserts!

What am I doing wrong (or is there an error in ```YAPP_Template_v3.scad```?

This is the complete code:
```
//-----------------------------------------------------------------------
// Yet Another Parameterized Projectbox generator
//
//  This is a box for Wemos D1 / bed Klokje
//
//  Version 3.3.5a (2025-02-24)
//-----------------------------------------------------------------------

include <./YAPPgenerator_v3.scad>

//-- which part(s) do you want to print?
printBaseShell        = true;
printLidShell         = true;
printSwitchExtenders  = true;
printDisplayClips     = true;

// ********************************************************************
// The Following will be used as the first element in the pbc array

//Defined here so you can define the "Main" PCB using these if wanted
pcbLength           = 100; // front to back (X axis)
pcbWidth            =  98; // side to side (Y axis)
pcbThickness        = 1.6; 
standoffHeight      = 3.0; //-- How much the PCB needs to be raised from the base to leave room for solderings and whatnot
standoffDiameter    = 7;
standoffPinDiameter = 2.4;
standoffHoleSlack   = 0.4;

//===================================================================
// *** PCBs ***
// Printed Circuit Boards
//-------------------------------------------------------------------
//  Default origin =  yappCoordPCB : yappCoordBoxInside[0,0,0]
//
//  Parameters:
//   Required:
//    p(0) = name
//    p(1) = length
//    p(2) = width
//    p(3) = posx
//    p(4) = posy
//    p(5) = Thickness
//    p(6) = standoff_Height = Height to bottom of PCB from the inside of the base
//             negative measures from inside of the lid to the top of the PCB
//    p(7) = standoff_Diameter
//    p(8) = standoff_PinDiameter
//   Optional:
//    p(9) = standoff_HoleSlack (default to 0.4)

pcb = 
[
  // Default Main PCB - DO NOT REMOVE the "Main" line.
  ["Main", pcbLength,pcbWidth, 0, 0, pcbThickness, standoffHeight, standoffDiameter, standoffPinDiameter, standoffHoleSlack]
];

//-------------------------------------------------------------------                            
//-- padding between pcb and inside wall
paddingFront        = 20;
paddingBack         =  2;
paddingRight        =  2;
paddingLeft         =  2;

//-- Edit these parameters for your own box dimensions
wallThickness       = 2.0;
basePlaneThickness  = 1.5;
lidPlaneThickness   = 1.5;

baseWallHeight      = 15;
lidWallHeight       = 10;

//-- ridge where base and lid off box can overlap
//-- Make sure this isn't less than lidWallHeight
ridgeHeight         = 5.0;
ridgeSlack          = 0.2;
roundRadius         = 3.0;

boxType             = 0; // Default type 0

// Set the layer height of your printer
printerLayerHeight  = 0.2;

//---------------------------
//--     C O N T R O L     --
//---------------------------
// -- Render --
renderQuality             = 8;          //-> from 1 to 32, Default = 8

// --Preview --
previewQuality            = 5;          //-> from 1 to 32, Default = 5
showSideBySide            = true;       //-> Default = true
onLidGap                  = 0;  // tip don't override to animate the lid opening
colorLid                  = "YellowGreen";   
alphaLid                  = 1;
colorBase                 = "BurlyWood";
alphaBase                 = 1;
hideLidWalls              = false;      //-> Remove the walls from the lid : only if preview and showSideBySide=true 
hideBaseWalls             = false;      //-> Remove the walls from the base : only if preview and showSideBySide=true  
showOrientation           = true;       //-> Show the Front/Back/Left/Right labels : only in preview
showPCB                   = false;      //-> Show the PCB in red : only in preview 
showSwitches              = false;      //-> Show the switches (for pushbuttons) : only in preview 
showButtonsDepressed      = false;      //-> Should the buttons in the Lid On view be in the pressed position
showOriginCoordBox        = false;      //-> Shows red bars representing the origin for yappCoordBox : only in preview 
showOriginCoordBoxInside  = false;      //-> Shows blue bars representing the origin for yappCoordBoxInside : only in preview 
showOriginCoordPCB        = false;      //-> Shows blue bars representing the origin for yappCoordBoxInside : only in preview 
showMarkersPCB            = false;      //-> Shows black bars corners of the PCB : only in preview 
showMarkersCenter         = false;      //-> Shows magenta bars along the centers of all faces  
inspectX                  = 0;          //-> 0=none (>0 from Back)
inspectY                  = 0;          //-> 0=none (>0 from Right)
inspectZ                  = 0;          //-> 0=none (>0 from Bottom)
inspectXfromBack          = true;       //-> View from the inspection cut foreward
inspectYfromLeft          = true;       //-> View from the inspection cut to the right
inspectZfromBottom        = true;       //-> View from the inspection cut up
//---------------------------
//--     C O N T R O L     --
//---------------------------

pcbStands =  
  [
  ];

//===================================================================
//  *** Connectors ***
//  Standoffs with hole through base and socket in lid for screw type connections.
//-------------------------------------------------------------------
//  Default origin = yappCoordBox: box[0,0,0]
//  
//  Parameters:
//   Required:
//    p(0) = posx
//    p(1) = posy
//    p(2) = pcbStandHeight
//    p(3) = screwDiameter
//    p(4) = screwHeadDiameter (don't forget to add extra for the fillet)
//    p(5) = insertDiameter
//    p(6) = outsideDiameter
//   Optional:
//    p(7) = PCB Gap : Default = -1 : Default for yappCoordPCB=pcbThickness, yappCoordBox=0
//    p(8) = filletRadius : Default = 0/Auto(0 = auto size)
//    n(a) = { <yappAllCorners>, yappFrontLeft | yappFrontRight | yappBackLeft | yappBackRight }
//    n(b) = { <yappCoordPCB> | yappCoordBox | yappCoordBoxInside }
//    n(c) = { yappNoFillet }
//    n(d) = { yappCountersink }
//    n(e) = [yappPCBName, "XXX"] : Specify a PCB. Defaults to [yappPCBName, "Main"]
//    n(f) = { yappThroughLid = changes the screwhole to the lid and the socket to the base}
//    n(g) = {yappSelfThreading} : Make the insert self threading specify the Screw Diameter in the insertDiameter
//-------------------------------------------------------------------
connectors   =
[
// 0, 1, 2, 3,   4,   5,   6,    7, 8, 9, ...
  [5, 5, 3, 3.2, 7.0, 4.0, 8.0, -1, 0, yappAllCorners
                                     , yappCountersink
                                     , yappSelfThreading] 
];

cutoutsBack = 
[
  [15, 5, 10, 10, 6, yappCircle, 0, 0, yappCoordBox] //-- use [0,0,0] of the box os origen
];

//===================================================================
//  *** Snap Joins ***
//-------------------------------------------------------------------
//  Default origin = yappCoordBox: box[0,0,0]
//
//  Parameters:
//   Required:
//    p(0) = posx | posy
//    p(1) = width
//    p(2) = { yappLeft | yappRight | yappFront | yappBack } : one or more
//   Optional:
//    n(a) = { <yappOrigin>, yappCenter }
//    n(b) = { yappSymmetric }
//    n(c) = { yappRectangle } == Make a diamond shape snap
//-------------------------------------------------------------------
snapJoins   =   
[
  [30,3, yappLeft, yappRight, yappFront, yappBack, yappSymmetric]
];

             

// **********************************************************
// **********************************************************
// **********************************************************
// *************** END OF TEMPLATE SECTION ******************
// **********************************************************
// **********************************************************
// **********************************************************

//---- This is where the magic happens ----
YAPPgenerate();
```

[image]

rosenhauer, 2025-08-20:
It looks like it was an issue with the template and generator getting out of sync.  

I've pushed a new update that should fix the error (param 7 is not the PCBGap it's the connector depth).  So you can remove the -1 (or change it to undef/yappDefault/default and it will work as intended. 

Sorry for getting them messed up.

mrWheel, 2025-08-20:
@rosenhauer 
Hi Dave,

Thanks for solving it so fast!

But, I now have this code:
```

//===================================================================
//  *** Connectors ***
//  Standoffs with hole through base and socket in lid for screw type connections.
//-------------------------------------------------------------------
//  Default origin = yappCoordPCB : pcb[0,0,0]
//  
//  Parameters:
//   Required:
//    p(0) = posx
//    p(1) = posy
//    p(2) = StandHeight : From specified origin 
//    p(3) = screwDiameter
//    p(4) = screwHeadDiameter (don't forget to add extra for the fillet or specify yappNoInternalFillet)
//    p(5) = insertDiameter
//    p(6) = outsideDiameter
//   Optional:
//    p(7) = insert Depth : default to entire connector
//    p(8) = PCB Gap : Default if yappCoordPCB then pcbThickness else 0
//    p(9) = filletRadius : Default = 0/Auto(0 = auto size)
//    n(a) = { yappAllCorners, yappFrontLeft | <yappBackLeft> | yappFrontRight | yappBackRight }
//    n(b) = { <yappCoordPCB> | yappCoordBox | yappCoordBoxInside }
//    n(c) = { yappNoFillet } : Don't add fillets
//    n(d) = { yappCountersink }
//    n(e) = [yappPCBName, "XXX"] : Specify a PCB. Defaults to [yappPCBName, "Main"]
//    n(f) = { yappThroughLid = changes the screwhole to the lid and the socket to the base}
//    n(g) = {yappSelfThreading} : Make the insert self threading specify the Screw Diameter in the insertDiameter
//    n(h) = { yappNoInternalFillet } : Don't add internal fillets (external fillets can still be added)
//-------------------------------------------------------------------
connectors   =
[
// 0, 1, 2, 3,   4,   5,   6,    7,  8, 9, ...
  [5, 5, 3, 3.2, 6.0, 4.0, 8.0,  7, -1, yappAllCorners, yappCoordPCB ]
];
``` 

And the Lid and Base Connecters are tide to each other.

If I code this (remove p(8)):
```
connectors   =
[
// 0, 1, 2, 3,   4,   5,   6,    7,  8, 9, ...
  [5, 5, 3, 3.2, 6.0, 4.0, 8.0,  7,    yappAllCorners, yappCoordPCB ]
];
``` 
... then there is room for the PCB between the Lid and Base Connectors.
I find this a bit confusing. Normaly "-1" represents the default value (or not?).

rosenhauer, 2025-08-21:
I was trying to get away from the arbitrary -1 for defaults as dinner
options -1 can be a valid value.  So it now looks for
undef/yappDefault/default.  The yappDefault was your suggestion.  That's
why I removed the -1 from the description.

Dave

mrWheel, 2025-08-21:
@rosenhauer 
Dear Dave,

Still a problem with the connectors.

The screw and screwhead are not going all the way through the base .... That is, in the pre-view.

[image]

After <F6> they are correct.

[image]

rosenhauer, 2025-08-21:
I've seen something similar on my old laptop where the interface between objects either appears when it shouldn't or comes and goes as the view is changed.   

I haven't seen it as much on my desktop or my new laptop but I also am running the development build of OpenSCAD from Jan 2025 (the first one that was released in 2025).  Not sure if that is the difference or maybe the default preview engine settings changed with either the version or the hardware change.  

But with your code from before. I'm not seeing the issue.

rosenhauer, 2025-08-24:
I'm on windows 11 so that may be part of the difference.  

I am using the default preview settings (as far as I can tell)

[image]

Not sure if the Mac has the same options.